gtagency/buzzmobile #1: Get Google Maps key Working

joshuamorton, 2016-09-16:
Add google maps key and keep it private so other people don't use it.

Also it currently exists in the git history, so we may want to go ahead and expunge it from history as well.

iRapha, 2016-09-16:
I already pushed changes to keep key private. Only things I need to do is expunge the key from the repo, correct?

iRapha, 2016-09-16:
I'm gonna have to temporarily disable continuous integrations so I can force overwrite the history to remove the keys

iRapha, 2016-09-16:
Yes. Also, correction. I'm not removing continuous integration. I'm removing master branch protection. (!!!!!!!)

diptodip, 2016-09-17:
rapha pls

iRapha, 2016-09-17:
friend this is not how you do.
after rewriting the history, you should have rebased, not merged.

closed 2016-09-17 by iRapha