gtagency/buzzmobile #16: Create and test joystick node

kpberry, 2016-10-12:


coletaylor788, 2016-10-19:
Need to revise the steering code to adjust it to the actual turning radius of the car. The joystick is too sensitive right now.

coletaylor788, 2016-10-21:
Code is pushed to actuator branch. Now should have full support. @iRapha I define a constant MAX_TURNING_ANGLE in the file to be 30 degrees. 1) does that sound about right? 2) Would it be better to make a global header file where we can store all these global constants? That way it would be easier to change if the hardware changes.

iRapha, 2016-10-21:
The `max_steering_angle` is already defined globally in the `constants.yaml`. You can use it as a rosparam:

`rosparam load buzzmobile/buzzmobile/constants.yaml`

Then in rospy you do: `rospy.get_param('max_steering_angle')` and in roscpp you do something like:

```cpp
float angle;
ros::param::get("max_steering_angle", angle);
```

As a note, currently, the value in `constants.yaml` is 1.0 rads. That is too large. I also think 0.52rad (30deg) is large. I think ~15 deg sounds more like reality, especially when you consider that this max angle is from the middle/forward angle of 0. So 15deg gives the wheels a range of 30deg to cover.

coletaylor788, 2016-10-26:
Still need to switch to constants.yaml

iRapha, 2016-10-26:
Oh, right, sorry. Lmk when you have a PR for that.

coletaylor788, 2016-11-01:
Closed by #93

closed 2016-11-01 by coletaylor788

iRapha, 2016-11-02:
Just a note that we also tested this :)