gtagency/buzzmobile #24: Implement obstacle break node

iRapha, 2016-10-16:


iRapha, 2016-10-18:
This involves:
- make `steering.py` send tentacle mask (with reasonable thickness) as ros msg.
- create `plan/break` node that...

- creates 'immediate future' mask, which is an arc at some distance from BM's position
- ANDs immediate future and tentacle masks together
- uses result as mask to `lidar_model`
- sums result to have final score
if score is greater than some threshold, send ros msg `/break` False
else True

Please note that 'reasonable thickness' might mean calculating how many pixels BM is wide (a measurement we do not currently have) and change the mask creation for that number of pixels.
(this might be a separate issue, but low-hanging fruit)

iRapha, 2016-10-19:
The list above is inacurate. Someone ping me about updating it tomorrow please

iRapha, 2016-10-19:
Updated the comment

coletaylor788, 2016-10-20:
@iRapha what do you mean by empty CarPose? Currently I am setting brake to true and velocity to 0. I am leaving the other inputs (steering angle and horn) as is. i.e. If you pressed brake and horn at the same time, it would brake and honk. Any concerns with that approach?

iRapha, 2016-10-20:
No concerns. That's exactly what I meant by empty

iRapha, 2016-10-21:
I started this on the `breaking` branch, sorry for the typo in the branch name. Todos include:

- debugging steering because the tentacle being picked it too long. I believe this has to do with the `travel_distance` rosparam being too large.
- test braking functionality with new image_consts

iRapha, 2016-10-21:
Just updated travel_distance, and that's fixed now. I'm gonna create some test images (or run the lidar bag file we have) and see how it does, then create a PR

iRapha, 2016-10-21:
Testing blocked by #55

iRapha, 2016-10-23:
Last break test to be run is using `some_lidar_data.bag`. This is blocked by #62.
Once that test is run, I'll make the PR.

PR #63 (2016-10-25): "Braking" and other fixes.
This PR...

- adds braking functionality to `steering.py` (fixes #24)
- fixes issures in `steering` wherein `sum(sum(img))` was overflowing, and essentially providing random scores to tentacles. For posterity: use `np.sum(np.sum(img))` instead.
- modifies architecture image for these changes, and for the CarState change happening in `actuators`
closed 2016-10-25 by iRapha