gtagency/buzzmobile #33: Formalize hardware features of car

iRapha, 2016-10-18:
This includes...
- measuring `wheel_base`
- re-measuring `wheel_circumference` (the rosparam we have was taken from buzzmobile-old)
- re-measuring `max_steering_angle`
- considering changes to the way we compute `travel_distance`

---

About the last point, right now, `travel_distance` is a value that is a "good enough" approximation of the distance travelled per tick. In buzzmobile-old, this was computed by the odometry interface, which assumed that the car was moving at a constant speed, and published:

`distance_travelled = (ticks_elapsed_since_last_keep_alive * wheel_circumference) / ticks_per_revolution`

We don't have odometry interface anymore, and we don't need it, because we're not simulating the car in the world model to make decisions about what to do when in a future state. So, to us, `travel_distance` can be an approximation because all it does it codify the granularity of the tentacle points. But I wanted to make sure this was mentioned somewhere ack'd

https://github.com/gtagency/buzzmobile/pull/32/files#diff-803bc1ddda718a341d80de27f0125f74R14

joshuamorton, 2016-10-18:
However, we do potentially have a complication because we (will soon) have brakes and those can make our real world speed not a direct function of our motor controller. It would be nice to have something (PID, Optical flow, a magnet attached to a tire, kalman filtering, etc.) to give us a real world feedback on speed so we don't have to guess.

Alternatively, as a stopgap, we can try and do something to lower our speed estimate when the brakes are being applied, but that is a hell of a lot of estimation.

iRapha, 2016-10-19:
I'd say we come up with an odometry solution to make our tentacle
predictions more accurate after mvp. Can you make issue about that maybe?

dacohen, 2016-10-23:
Front Axle to Rear Axle Distance: 2.032 m
Left Wheel to Right Wheel Distance: 1.3716 m

iRapha, 2016-11-01:
steering angle eyeballed to about 15 deg

iRapha, 2016-11-02:
road width ~8feet (needs conversion to meters)

iRapha, 2016-11-02:
min_fix_distance can probably be bigger than 0.01 meters (that might be in gps' margin of error)

iRapha, 2016-11-02:
travel_distance will be calculated with odomcallback #41

iRapha, 2016-11-02:
`breaking_distance` will be based on odometry too #54

iRapha, 2016-11-02:
`buzzmobile_width` ~= 66 in (convert to meters)

PR #118 (2016-11-09): Hardware measurements
Closes #33. Related issue involving `travel_distance` (#41) still open.
closed 2016-11-09 by williamsam1