gtagency/buzzmobile #38: Discuss acquisition of dedicated car laptop

iRapha, 2016-10-19:
@joshuamorton @dacohen please chime in

joshuamorton, 2016-10-19:
For reference, RJ can't give us theirs after some discussion.

joshuamorton, 2016-10-24:
This discussion should be partially tabled until next semester, and we may want some input from Matt/RJ in terms of what some good low power or self-poweered computing hardware options would be, or if we should just requisition a laptop. Otherwise, we can probably get something budgeted in the Spring maybe.

iRapha, 2016-10-24:
I guess all our tests will have to run in the shitty VMs then xD

joshuamorton, 2016-10-25:
For now :/

jgkamat, 2017-03-28:
@iRapha I would also reccomend looking into surplus. They might not have laptops though, but I do know they have optiplexes you might be able to take.

iRapha, 2017-03-28:
@jgkamat Will def look into it! The reason we wanted a laptop is because we still wanted to be able to do hot-fixes on the car itself. Thanks for the info tho.