gtagency/buzzmobile #39: Prune repo of unused things

iRapha, 2016-10-19:
Such as two of the `image_const` nodes (we only need one, and it should arguably live in `tools`).
Or even the `lane_detector`, which we are scraping for the future `road_detector` node aka agencynet

joshuamorton, 2016-10-19:
Ehh, I agree with the first part of this (redundant stuff), but the whole point of buzzmobile was to potentially have various ways of doing things, and use it as a testbed for stuff.

iRapha, 2016-10-19:
I'd argue those can happen in branches but master should be cleaner

joshuamorton, 2016-10-19:
Depends, there's value in being able to test multiple things at once,
though I guess if you're smart enough with git you can merge non-branches
into each other to get that effect. Worth weighing in mind in any case.

closed 2016-10-21 by iRapha