gtagency/buzzmobile #41: Implement Real world Odometry

joshuamorton, 2016-10-19:
Following discussion in #33, we need some way to get real world motion to help with planning.

iRapha, 2016-10-21:
This will probably be much more easily done through some hardware that gives us a callback. Pending discussions.

iRapha, 2016-11-02:
So I *believe* this works, but we weren't able to test it yesterday due to time constraints and other issues. Therefore I'm keeping the issue open until we have it fully tested. Might be even easier when we have merged #100

iRapha, 2016-11-02:
Once we have tested it, we can work on the issues that *use* the odometry to do things

iRapha, 2016-11-09:
So we have this in the same place, this comment was made in issue #33:

`travel_distance` will be calculated with odomcallback #41 
`num_points_in_tentacle` will be updated to use that ^ correctly, or can be done to say "total travel needs to be 6 meters, so do as many points as needed, except have a max number of points"

iRapha, 2016-11-13:
We tried outputting what odometry callback published and we're getting consistent 0s. I think something might be off either hardware or software-wise. we need to look into it.

iRapha, 2016-11-22:
PR #146 temporarily disables odom callback through Serial port. The reason being that we were sending those callbacks at every command received, which was blocking the Serial port half the time. The solution is to send odom callbacks through serial only once every x milliseconds.

iRapha, 2016-11-30:
PR #150 reenables odom callback, and fixes it altogether. String formatting in the .ino files were the initial cause of pain, and reading form Serial does not cause blocking issues.