gtagency/buzzmobile #43: Fix/Examine Chain for the Motors on the Car

coletaylor788, 2016-10-19:
The chain makes clicking noises occasionally as driving straight and clicks very quickly when turning sharply. It doesn't really sound right.

iRapha, 2016-10-19:
The situation on this seems to be that, the way the chain is currently attached, the car cannot move faster or slower than what the motor is moving at, otherwise pressure is put on the motors by the chains (it's a good thing the chains will break before the motor does).

So the fix for this is to add a 1-way coupling on the rear-axle. But this means the car won't be able to go on reverse. So discussion on this needs to happen. In any case, something needs to be done, otherwise driving will be finicky.

I'd also want to check with hardware team if the fix for this can happen before the weekend RJ will take the car away, mainly because we'd like to do individual systems' tests without fear of breaking the chain. If it's not possible, our alternative is to lift the car with a jack and do stationary tests (the motor moves wheels, but car does not move because wheels are in the air)

iRapha, 2016-10-26:
I'm gonna rename this bug, and create a new one for tracking the fact that we have no brakes ( #79 )

iRapha, 2016-10-29:
After we have this we can add a `NEUTRAL` CarState so we can push the car without fear of breaking chain or motor.