gtagency/buzzmobile #46: Fix if statement for estop in Arduino code

coletaylor788, 2016-10-19:
if(digitalRead(estop_pin) == HIGH) is never evaluating to true, even when the estops are disabled. Since this never evaluates to true, the car never moves. Commenting this if statement allows the car to run but it was likely important and should be fixed. Here is the comment from the code:

/* Only run the controllers if the motors are enabled.
   * This prevents the controllers ramping up while the
   * vehicle is estopped, creating a dangerous situation.
   */

iRapha, 2016-10-19:
For more context, this e-stop seems to be connected to serial port 6 on the arduino, which we were unable to trace to a hardware switch. I believe there exists one though, we need to check again

iRapha, 2016-10-26:
I believe you fixed this in #31, right, @coletaylor788?

coletaylor788, 2016-10-26:
No. I un-commented the code but the problem is still there. Either the if statement is checking the wrong thing or there is some hardware problem where that pin is not getting a HIGH signal when the e-stops are disabled.

iRapha, 2016-10-26:
We have a separate issue ( #36 ) for tracking the hardware problem. This issue is just for the code.

coletaylor788, 2016-10-26:
Okay. I'm not sure if it is a hardware or a code issue. If it isn't a code issue, then this could be closed.

iRapha, 2016-10-26:
We are assuming it's a hardware issue. If when we close #36 we find nothing wrong with hardware, we will reopen this.

closed 2016-10-26 by iRapha