gtagency/buzzmobile #48: Consider removing horn_node

iRapha, 2016-10-19:
Currently, we have no horn. But with the assumption that we will get one (this is pending discussion):

Shouldn't `car_interface` also process `CarPose` for handling turning horn on and off? Why do we need a separate node and separate message simply to say "horn is on/off", when the `CarPose` will already encode this information? (and if it doesn't currently encode it, it should shouldn't it?)

coletaylor788, 2016-10-19:
I agree, it seems unnecessary. It is just that car_horn was a node ported from buzzmobile_old. Not sure if there was some reason for breaking it out separately last time.

iRapha, 2016-10-26:
@coletaylor788 didn't you fix this also?

coletaylor788, 2016-10-26:
Probably fixed but not tested

iRapha, 2016-10-26:
We'll we can't actually test the horn at all anytime soon because we won't have a physical horn (issue tracked in #66 ). I'll close this issue and whenever we get a horn I'll make a new issue to check implementation.

closed 2016-10-26 by iRapha