gtagency/buzzmobile #61: Make setup.py file in project root

kpberry, 2016-10-23:


joshuamorton, 2016-10-24:
I started this on a (now deleted) branch. I can push it again as a starting point if you'd like.

Also what's the goal of setup.py in this case

iRapha, 2016-10-24:
The goal is to install python deps automatically. Sorry if I deleted your branch, I thought I was only deleting stale branches. In any case, @kpberry got started on this with some of our real deps in https://github.com/gtagency/buzzmobile/launch

joshuamorton, 2016-10-24:
No, I'm pretty sure I deleted it a while back (and my goal was for a ros package, so different goals there I think).

joshuamorton, 2016-10-24:
On second thought, can we (either here or in another issue) change this to "automate the deployment process" such that we factor out everything into an install.sh that runs setup.py, etc. and so that you can git clone and ./install.sh and run. This way we have a combined deploy process across CI, live environment, and dev environments, that's really easy to use.

iRapha, 2016-10-24:
What do you mean a ros package? We still have `package.xml`, but that's not a python thing. Oh do you mean that the goal was to make all of buzzmobile one package so that relative imports would work? I don't think we need that anymore because there aren't shared deps, and shared constants are rosparams now.

I agree that deployment needs to get better. Can you add a low-priority issue for Spring Carnival?

joshuamorton, 2016-10-25:
Exactly, that's what I mean. And yea.

closed 2016-10-26 by iRapha