gtagency/buzzmobile #64: Make deployment similar on all platforms

joshuamorton, 2016-10-24:
Following some discussion in #61, it might be better if we factored out parts of the travis ci build script into a `install.sh`, so that 

    git clone
    ./install.sh

would install system dependencies, run `setup.py`, install rosdeps, etc. and allow deployment to be done the same way across all systems.

joshuamorton, 2016-10-26:
I've started on this, basic workflow will be

 - run `install.sh`
   - runs `python3 -m venv buzzmobile`, creating a virtualenv
   - install tensorflow from wheel
   - install opencv from source
   - install the ros dependencies
   - run setup.py
     - install all the python dependencies
   - run `rosdep`
   - run tests?

Current biggest issue is getting OpenCV to work (efficiently), because it doesn't play well with Ros or python3 or virtualenvs on their own, and we're using all three.

iRapha, 2016-10-26:
Hm. travis is currently doing most of those, including opencv. For a fuller list of things `install.sh` should do, maybe you can look at `.travis.yml`?

PR #90 (2016-10-29): travis -> DoCIF
Changes the continuous integration tool used by our repo from travis ci to DoCIF, which uses circle-ci, docker, and virtualenv to allow us to have multiple tests and statused per build run.

Fixes #88 
Fixes #64 
Fixes #89
Part of #87 
Part of #51 (along with #86)
closed 2016-10-29 by iRapha