gtagency/buzzmobile #102: Car State goes from START to MANUAL

iRapha, 2016-11-02:
I'm pretty sure this happened last night. Worth doing some more tests about. If it is actually happening, we might want to look at https://github.com/gtagency/buzzmobile/blob/master/buzzmobile/plan/controller/src/controller.cpp#L82 . This `if` might be failing for one reason or another.

coletaylor788, 2016-11-08:
After starting the nodes manually, the car state appears to function correctly. We should test this again next time we test on the car and look at the "car_state" message as we test.

iRapha, 2016-11-13:
Tested again. I think something in the way we were launching was not letting this work. In either case, I'm pretty sure the last kinks of this issue are fixed in #132.

closed 2016-11-22 by iRapha