gtagency/buzzmobile #112: Implement Simulation Using ROS and Gazebo

sastorer, 2016-11-04:
- [ ] Integrate Gazebo packages into catkin
- [ ] Add URDF files
- [ ] Get CAD from RJ for visual elements of the simulation

You should be able to set it up without anything visual and have everything work just fine.

Resources:
[ROS Wiki](http://wiki.ros.org/gazebo_ros_pkgs)
[Actual Gazebo Website](http://gazebosim.org/)
[URDF and Gazebo Docs](http://gazebosim.org/tutorials?tut=ros_urdf&cat=connect_ros)

iRapha, 2016-11-04:
This is a partial duplicate of #20 but I'll let Josh's PR ( #111 ) close that issue and keep this one because you have checkmarks

sastorer, 2016-11-04:
Ah, I looked to see if there would be any duplicates. I missed that one because it was buried under other things that that issue covers. We can keep track of just Gazebo things here for greater clarity.

iRapha, 2016-11-09:
@dacohen is working on making URDF files

Daniel, you might want to look at [1] and [2] for some ready-made interface for some ackerman's steering models. I can't find ready-made URDF for them, but I'm sure they exist.

[1] http://wiki.ros.org/Ackermann%20Group

[2] http://docs.ros.org/api/ackermann_msgs/html/msg/AckermannDrive.html

joshuamorton, 2017-02-16:
#181, how is this going?

iRapha, 2017-02-16:
According to @evanxq the actuators work (from car pose to simulation), which means that #182, #160, #159, will all be fixed with PR #181, which is in code review rn. Left on this will be: #161, #179 #158, but the last two are low priority

iRapha, 2017-02-19:
For future reference, implementing simulation tests in pytest might not be as straight forward as `@with_launch_file('buzzmobile', 'simulation.launch')`. See [0]. "gazebo_rosmaster_uri is also a thing", says @joshuamorton. Might require a whole new set of test utils. 

[0] http://answers.ros.org/question/193062/how-to-run-multiple-independent-gazebo-instances-on-the-same-machine/

joshuamorton, 2017-08-20:
So from the investigation I did yesterday, here's what (I believe) the state of gazebo simulation is:

1. Some dependencies aren't installed by default, this is easy to fix with a `rosdep install`, but you need to know which things they are. For reference, I believe its the needed packages are `hector_gazebo_plugins controller_manager controller_manager_msgs`, at least that's what seemed to work on my system
2. This is pseudo-blocked by #189. I think a minimum viable product here would be a node that publishes a constant polyline. In the make-sim-work branch, I'm currently just not running any of the googlemaps nodes, which is a fine temp workaround.
3. Even with that, there appear to be some issues with simulation. I get errors (
[see here](https://github.com/gtagency/buzzmobile/files/1237377/temp.txt)) that point to a misconfiguration in the URDF or joint mapping, I think.

@iRapha @chsahit

iRapha, 2017-08-20:
I'm gonna throw someone under the bus here and tag @kpberry bc he said he wants to help car this semester