gtagency/buzzmobile #113: Investigate test flakyness

joshuamorton, 2016-11-04:
When writing ROStests, it is necessary to spin up roscore and ros nodes and have them communicate. This requires multiple system threads to run in sync. There can be race conditions, timing errors, and deadlocks as a result of various communication issues with this. It would be nice if we developed some tooling to prevent/mitigate that kind of problem.

iRapha, 2016-11-04:
Does catkin_make test handle part of this? Should we consider going through the hassle of setting up catkin tests?

joshuamorton, 2016-11-04:
I don't see how it could, it just runs the test code. Their examples run
with catkin_make include sleeping in the python code. That's where the
flakyness lies.

iRapha, 2016-11-04:
Actually part of it would fix flakyness, just not the catkin_make part.

What we need to do is put the test nodes in a rostest file, which is similar to a roslaunch file. Then test using the `rostest` command, not the `pytest`. This guarantees that roscore will be running (because it works just like a roslaunch command).

joshuamorton, 2016-11-04:
Roscore running or not isn't affecting flatness, the issue is spinning up
test nodes inside the Python code when actually running the tests.

iRapha, 2016-11-04:
Hm. If you're talking about `node = roslaunch.core.Node('buzzmobile', 'inputer.py')` I thought you could them up in the rostest file, and set a param to respin it up for each test. Basically let rostest handle that for you. I can't find that a param like that exists, so an alternative is to do `assert rosnode.rosnode_ping("inputer", max_count=5)`, to at the very least see if the node went up. It doesn't fix flakyness but it might help with debugging it.

In either case, doing a rostest file is a more elegant solution than `sleep 2` in hopes roscore is done, and then killing it by PID.

joshuamorton, 2016-11-04:
I'm going to investigate a couple things

1. spinning up roscore within python for each test
2. Check for roscore to be running before each test begins
3. some utility methods to do things like a thirty second timeout, but with callbacks that will end the test as soon as it is done
4. Some wrappers to make sure nodes are shut down and such correctly.

iRapha, 2016-11-04:
My main concern is whether these exist and if we can reuse them

joshuamorton, 2016-11-04:
Well, 2 is like a 1 liners, so more or less yes, we just need to use it, 1 maybe does, but its also low hanging fruit. 3 might exist, and 4 doesn't from what I've seen.

The other problem with using catkin to test is that you seem to lose out on any kind of coverage stats.

PR #121 (2016-11-28): Rostest
Please tell me what to explain and comment on.

Fixes #113
closed 2016-11-28 by joshuamorton