gtagency/buzzmobile #126: Make install script failsafe

iRapha, 2016-11-12:
If it fails, it's hard to run it again after fixing the issue that made it fail the first time.

The issues seem to be the catkin_init and rosdep init. They fail if the've already been run.

iRapha, 2016-11-13:
as a note, cole said it seems to be failing when apt-get is run sometimes, because ubuntu might be running apt-get update in the background, which puts a lock in it.

possible solution, tho jank: https://askubuntu.com/questions/132059/how-to-make-a-package-manager-wait-if-another-instance-of-apt-is-running

iRapha, 2016-11-13:
To test this: create new VM, install buzzmobile, try running a launch file.

joshuamorton, 2016-11-13:
Alternatively, use `wait` in the install script, this will block until all background processes are finished.

iRapha, 2016-11-13:
bash files are async???

joshuamorton, 2016-11-13:
I'm gonna to with 'maybe', I don't actually think it should be in this case, but if that's the problem, this should be a fix. Unless you mean that the system is auto-updating, but I don't think that's a thing that happens.

iRapha, 2016-11-13:
Actually I do think it happens. From what we read while trying to fix this
issue, it seems ubuntu might start up apt-get update (or something else
that locks apt-get) at a random point and that can cause the issue we found.

joshuamorton, 2016-11-13:
Gross, I didn't find that when I was looking around.

iRapha, 2016-11-13:
This is the main one we looked at, because the error is the same as what we got:

https://stackoverflow.com/questions/23610628/detecting-avoiding-multiple-apt-get-dpkg-installs

> It works great, except in the case where some system process kicks in and starts updating APT databases in the background.. Then it fails with dpkg: error: dpkg status database is locked by another process.

joshuamorton, 2016-11-13:
Keep in mind that that fix requires apt-get installing the tool.

iRapha, 2016-11-13:
Sure, I wasn't suggesting we use the solution in this stackoverflow. I was just showing you that it's most likely that our issue came from a background apt-get process updating databases.

coletaylor788, 2016-12-05:
As an update I edited the install script to use a wrapper function for apt-get that waits for other apt-get processes to terminate. It appears to be not fail due to apt-get anymore. However, the application still can't be run after preparing the environment this way. I will continue looking into it.

joshuamorton, 2016-12-05:
What issue are you hitting after build (on a clean env I assume?)

iRapha, 2016-12-05:
Sg

coletaylor788, 2016-12-05:
@joshuamorton I can't remember exactly (was about a week ago) but I believe they looked like python dependencies.

coletaylor788, 2016-12-05:
I plan on taking another look in the next couple of days to get more information

coletaylor788, 2016-12-07:
Here is a couple of lines from the error that I think summarizes it.

```
ImportError: Could not find Qt binding (looked for: 'pyqt', 'pyside'):

ImportError for 'pyqt': No module named 'sip'
ImportError for 'pyside': No module named 'PySide'
```

So it looks like it is looking for pyqt or pyside as the QT binding for RQT (used for mission control). pyqt has a dependency on sip which doesn't appear to be installed and pyside is also missing.

Do you know anything about this and do either of you care which binding is used? It looks like pyqt is the default binding which would probably be easier to fix.

coletaylor788, 2016-12-07:
@iRapha @joshuamorton

iRapha, 2016-12-07:
Is rqt still failing to install? We might need to look up how to properly
install it. Maybe there's an rqt_core or something we need to be depending
on.

coletaylor788, 2016-12-07:
Yes, the problem is with rqt. I can look further into how to get it installed but want to make sure our app isn't depending on PySide vs. PyQT. Is there one that needs to be used as the binding or is either fine?

iRapha, 2016-12-07:
I believe neither of those are being explicitly used by us.

coletaylor788, 2016-12-07:
Okay, thanks. I will proceed with the easiest way to install rqt then

coletaylor788, 2016-12-09:
As an update, the RQT problems only occur in the virtual environment. If you run it outside the virtual environment, the issue doesn't exist.

iRapha, 2016-12-09:
This probably means that whatever is installing rqt is doing so outside of the venv.

coletaylor788, 2016-12-12:
As an update, I think the issue may be related to Python 2 vs. 3. If I enter a python3 shell outside of the virtualenv and enter: "from qt_gui.plugin import Plugin", it gives the same import error (Sip and PySide). Howerver, it works from within the python (2) shell.

coletaylor788, 2016-12-12:
Additionally, the import line works in python2 in the virtualenv.

coletaylor788, 2016-12-12:
@iRapha @joshuamorton Are we supposed to be using Python 2 or 3?

iRapha, 2016-12-12:
Python3. As soon as the install script is confirmed working, we'll all get new VMs and start using python3 only. So whatever is installing pyqt is installing it for py2, which is not what we want. Maybe it's like a flag we need to set for rosdep? Or maybe we need to depend on a different version of rqt so it knows we're using py3?

coletaylor788, 2016-12-12:
Status update at end of night:

When run with python3, it says it is missing sip. After manually compiling and installing sip with python3 (pip can't install it), it said it was missing PyQt. I installed and compiled it from source as well which removed that error. Now the error is "basestring is not defined". basestring is something only defined in Python2. It starts with rqt_gui which calls a method in qt_gui which is in python2.7.

iRapha, 2016-12-19:
So there's a slight chance we are going to have to move things to python2 (which should only be a matter of changing install script, since all the code has been locally tested in py2 anyways). I'm deferring this to @joshuamorton pls dont abandon us pls

joshuamorton, 2016-12-19:
Indeed, it should be mostly issues in the install script, everything else should be backwards compatible (also `yield from` isn't supported in 2.7).

coletaylor788, 2016-12-20:
Just let me know when y'all decide which version we plan on using. If we go with python 2, I will update the script as it should be easy to get working. Otherwise, once it is decided, I can keep troubleshooting the issues with python 3. There seems to be incompatibilities with rqt and python 3 where fixing one issue is just revealing another.

joshuamorton, 2016-12-20:
Do py2, it'll be easier.

I hate ros.

coletaylor788, 2016-12-20:
Okay, I'll work on it when I get time.

PR #155 (2017-01-16): Install script
@coletaylor788 Let me know if this currently does everything you want it to.

If so, this...
Closes #152 
Closes #126
closed 2017-01-16 by joshuamorton