gtagency/buzzmobile #140: make controller brake like steering does

iRapha, 2016-11-17:
Whenever steering wants to brake, it just sets the `pose.brake = True`.

When controller wants to brake, it also sets velocity to be negative.

This issue includes:

[] verifying that car_interface is properly braking when `pose.brake` is set, and velocity is null.
[] making controller's brake interface better, by having a dedicated "brake" button, that sets `pose.brake` to true. AKA: I don't want to have to reverse in order to brake the car :)

coletaylor788, 2016-11-22:
@iRapha The controller node should have a brake button that `sets brake = True` and sets `velocity = 0`. I think the reason we were having to use reverse to brake is because we were actually setting velocity. Since there aren't any brakes on the car and we were setting acceleration to 0, it would just coast to a stop. Let me know if I am missing something.

iRapha, 2016-11-22:
So I agree with you that's how things should be working, but I think that we need to handle setting the velocity in `car_interface`.

`controller` and `steering` should not care about the hardware details. They should be able to just set `brake = True` and the details about whether we have breaks and are activating those or we don't have brakes and have to set `vel=-maxspeed`. Handling this should happen in `car_interface`.

The first part of the above paragraph seems to be the way things work currently, given [this](https://github.com/gtagency/buzzmobile/blob/master/buzzmobile/plan/controller/src/controller.cpp#L123).

So this issue is just a matter of making sure that `car_interface` is translating a "break" message into setting `velocity=-maxspeed`, which is [doesn't seem to currently do](https://github.com/gtagency/buzzmobile/blob/master/buzzmobile/drive/car_interface/src/main.cpp#L69).

Then one day when we get brakes ( #79 ) we can make `car_interface` just activate those.

Furthermore, the readme does not include what button is the break button.

Then we just need to test that braking works in general (from controller, by pressing brake button, and from steering by sending `brake=True` pose message).