gtagency/buzzmobile #149: Multiple Launch files is not currently support in testing

joshuamorton, 2016-11-25:
Currently, multiple trying to use `rostest_utils.with_launch_file` for multiple launch files will cause silent errors.

iRapha, 2016-11-26:
TODO is here: https://github.com/gtagency/buzzmobile/pull/121/files#diff-0b78e1468bd4bce4d2aec5aea89987beR49

iRapha, 2017-02-16:
@joshuamorton correct me if I'm wrong but isn't this almost fixed with #180 too?

joshuamorton, 2017-02-16:
I don't think I do much to address this.

The issue is still that the @with_launch_file decorator will override parent launch files.

iRapha, 2017-02-16:
Oh ok. Honestly I'm considering closing this issue until we actually express a need for loading multiple launch files...

joshuamorton, 2017-02-16:
Given that afaik we can have launch files import other launch files, I
don't see a strong need for it either.

closed 2017-02-16 by iRapha