gtagency/buzzmobile #157: Add arg support for test_utils

iRapha, 2017-02-10:
Launch files can take optional args:

```xml
<arg name="gzclient" default="false"/>

<group if="$(arg gzclient)">
    <node pkg="gazebo_ros" name="gazebo_gui" type="gzclient"/>
</group>
```

These can be set via launch files that include other launch files:
```xml
<include file="$(find buzzmobile)/launch/simulation.launch">
    <arg name="gzclient" value="true" />
</include>
```

And via command line:
```bash
$ roslaunch simulation.launch gzclient:=true
```

We'd like a way to set these in the test framework too:
```python
@with_launch_file('buzzmobile', 'simulation.launch')
# should also allow:
@with_launch_file('buzzmobile', 'simulation.launch', gzclient=True, world='something.world')
```

joshuamorton, 2017-02-11:
Some background:

the roslaunch script is housed in `/opt/ros/indigo/lib/python2.7/dist-packages/`. Within that, 

 - `roslaunch/__init__.py` contains the main script that runs everything
 - this calls out to `roslaunch/rlutil.py` a lot, including some parsing functions
 - one of these calls out to `rosgraph/__init__.py` which decides to delete all the arg strings before anything is done with them.

If someone wants to, I'd suggest starting by adding the line `import pdb; pdb.set_trace()` on line 221 of `roslaunch/__init__.py` and running then running a real example of a roslaunch that uses a set of flags. Then stepping through and seeing what happens.

In a toy example, I got no useful results.

joshuamorton, 2017-02-15:
Oh also my poking around yesterday let me figure out how to do this.

joshuamorton, 2017-02-16:
(perhaps the easiest way is to update `sys.argv`, since we really don't care about it.

iRapha, 2017-02-17:
I think the key to this is the `roslaunch_strs` which can be passed in to `ROSLaunchParent`, who then "loads" them using `roslaunch.config.load_config_default` which parses the strings as XML.

I'm trying this out real quick to see if it works

PR #184 (2017-02-19): Enable args in with_launch_file
Not ready yet!

Closes #157
closed 2017-02-19 by joshuamorton