gtagency/buzzmobile #177: Figure out how to run pytests in parallel

iRapha, 2017-02-13:
Right now, everything is single-threaded and boxed because things can interfere with each other through global roscore processes otherwise.

iRapha, 2017-02-13:
Note that simulation tests are not yet running in boxed environments. We might need to do so when we have multiple test files.

joshuamorton, 2017-02-13:
So throwing some thoughts out.

Currently in `test_utils/mock_nodes`, the listener node is created and then we do a crappy attempt to tear it down at each iteration, this sometimes works, but not always.

Another option is to create a single "test_listener_node". Then, this listener subscribes and unsubscribes to topics for each test, but the *node* is essentially a singleton that self-modifies instead of being recreated. This is a bit more stuff to self manage, but not a whole lot more.

This also allows multiple listeners in multiple threads, meaning that you can run this in parallel pretty easily, as long as you prepend a test namespace per test, even on a single roscore.

The other thing we can try to do is create multiple roscores, but that is still proving difficult.

joshuamorton, 2017-02-13:
update: I think I know how to spawn multiple roscores, finally!!!

(this also allows multiple gazebo's in parallel)

closed 2017-02-17 by iRapha