gtagency/buzzmobile #204: Measure the max speed of the car and caluclate max torque.

chsahit, 2017-05-11:
The first one can be done trivially enough on a track with a stopwatch. The second one can be done by looking at the datasheets for the CIMs and talking to robojackets about the gear ratio.

iRapha, 2017-05-11:
One thing to note is that we *don't* want the `max_speed` rosparam to be the maximum speed the hardware can reach. We want it to be the maximum allowed speed for the car to run. This is why I don't see why a value like 0.2 is that bad (though I see why it is bad for simulation in your computer lmao).

I'd suggest that, instead of measuring the max_speed, you just make the car run at a certain speed that is slow enough for it to not run people over in a parade and make that `max_speed`. That's how we arrived at 0.2 but you can probably increase that.

chsahit, 2017-05-13:
Ok so as a note, the max torque, which does need to be measured can be found at:
` buzzmobile/simulation/sim_car_interface/src/sim_car_interface.cpp :43`

iRapha, 2017-05-13:
Don't forget to also put this as an inline comment in `constants.yaml`. Because the whole point is that if we're gonna have to maintain that being in two places, we better *know* that it needs to be modified in two places everytime we do modify in one of the two.

chsahit, 2017-05-13:
Should I also make another issue that both places need to be updated? Because that's a definite bug.

iRapha, 2017-05-13:
yes

chsahit, 2017-06-17:
So if we don't want the max speed/torque to be what the car's capabilities, this issue is resolved by deciding on a max speed, not measuring one right?

iRapha, 2017-06-20:
Correct