gtagency/buzzmobile #213: Move to ROS Kinetic

chsahit, 2017-11-15:
Most things in bzm are ROS distro invariant as I could run a lot of the nodes on my computer without running a VM. The install script will have to be ported as spinning up the nodes in the bzm package to make sure we're not using deprecated standards and no dependencies are missing. The CI instance will also now have to run Ubuntu 16.04

chsahit, 2017-11-15:
@joshuamorton will this affect pyrostest? If so I won't keep pushing on this one until we iron out sim issues.

jgkamat, 2017-11-15:
Re testing, you could steal the good bits from https://github.com/gtagency/buzzmobile/pull/202 and just change the `FROM` in the dockerfile from `indigo` to kinetic`, which is an ez upgrade :)

joshuamorton, 2017-11-15:
It shouldn't. Try it and I'll fix any problems.